lubeda/EsphoMaTrix #10: LED doesn

YujiTFD, 2022-05-11:


jd1, 2022-05-11:
Try to start without this component and configure only light with 'platform: neopixelbus'. If this also doesn't work, check your wiring.

closed 2022-05-11 by lubeda

YujiTFD, 2022-05-11:
Yes, I got it working with monobit after which bitmap cozette worked too, not sure what was causing problem. Still, cozette 13px looks trancated from the top, like this (I sent on screen "123", don't mind reflection). Is it something amendable?
[image]

lubeda, 2022-05-13:
You can scale the font with the size parameter in the [font definition](https://esphome.io/components/display/index.html?highlight=size#fonts). But the font is not readable when scaled to fit!