lubeda/EsphoMaTrix #13: time format and date format

kelvincabaldo07, 2022-06-01:
Hi,

First of all, thank you for this awesome project of yours! Im using it at the moment and it is fantastic! If i may request, is it possible to have as a setting the hour and date formats. Say 12 hour format for the hours or the inclusion of seconds and for the dates, the inclusion of year and changing to different locale. 

Thanks!

kelvincabaldo07, 2022-06-03:
> There is a new branch "date-time" which offers new parameters to configure the display format.
> 
> I think these are fine for USA:
> 
> ```
> ehmtx:
>   time_format: "%I:%M%p"
>   date_format: "%m.%d."
> ```
> 
> Without setting these parameters it will show the european style. Now there is no scrolling in case of a long text!!!
> 
> The strftime format is described here https://esphome.io/components/time.html?highlight=strftime
> 
> give it a try and report the results so i can merge it

this works great! thanks! i will add some other feature request if you dont mind. :)

lubeda, 2022-06-04:
will be merged after some testing

closed 2022-06-04 by lubeda