lubeda/EsphoMaTrix #19: Time / date switching does not work

pepe59, 2022-06-25:
Time / date switching does not work after updating ESPhome. I tried to add:
```
  show_date: true
  date_format: "%d.%m."
```
That didn't help. Only the time is still displayed.

lubeda, 2022-06-25:
Try 
`show_screen: 14` and `show_clock: 7` With this values the data should be displayed for` show_screen - show_clock seconds` in this example 7 seconds. If it still doesn't work. you have to wait because I am far away from home on holiday for 2 weeks.

pepe59, 2022-06-25:
It doesn't work in this setting either, but thank you.
There is no place to hurry. Enjoy a nice holiday.

lubeda, 2022-07-11:
can you try the new "2022.7.0" branch
```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EsphoMaTrix
      ref: "2022.7.0"
    refresh: 60s 
    components: [ ehmtx ]  
```

pepe59, 2022-07-11:
I confirm, it works perfectly.
I also tried changing the time.
```
ehmtx:
  id: rgb8x32
  show_clock: 10
  show_screen: 15
```
All without a problem.
Thank you very much.
If you don't need to test anything else, we can close the issue.

closed 2022-07-12 by pepe59