lubeda/EsphoMaTrix #29: can I use this for a 44x11 pixel matrix

teitelbot, 2023-01-19:
I bought the higher resolution matrix and I thought it would just be a matter of changing the width and height in the code but when I do that it just crashes the whole thing. Am I doing this wrong?

lubeda, 2023-01-19:
Hi,
there are multiple possibilities. One is the limited RAM of the ESP. Other reason might be the, more or less hardcoded, geometry of the display. I think it may be possible to adapt the code but it is not that easy.  Can you provide a link to the used led matrix?

teitelbot, 2023-01-19:
This is the product I bought
https://amzn.to/3CZCWuw

teitelbot, 2023-01-23:
@lubeda What esp should I purchase to increase RAM? Im using the d1mini now.

lubeda, 2023-02-14:
I think it would be possible to make my code work with your screen layout. But e.g. the common icons you get on the internet are all 8x8. I can't spend my time on this feature request. But pull-request are welcome.

closed 2023-02-14 by lubeda