lubeda/EsphoMaTrix #31: not work gif icons

jrspowers, 2023-02-07:
hello i started today with your nice project everyone i run into 1 big problem with all icons with gif don't work i can't see any movement what am i doing wrong see configuration?

```yaml
substitutions:
  devicename: esphomatrix
  mypin: GPIO4
  board: nodemcuv2
  
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EsphoMaTrix

esphome:
  name: $devicename
  on_boot:
    priority: -100
    then: 
      - ehmtx.text.color:
          id: rgb8x32
          red: !lambda return 200;
          blue: !lambda return 170;
      - ehmtx.today.color:
          id: rgb8x32
          red: !lambda return 10;
          green: !lambda return 250;
      - ehmtx.clock.color:
          id: rgb8x32
          red: !lambda return 50;
          green: !lambda return 150;
          blue: !lambda return 230;
      - ehmtx.alarm.color:
          id: rgb8x32
          red: !lambda return 250;
          green: !lambda return 50;
          blue: !lambda return 30;
      - ehmtx.weekday.color:
          id: rgb8x32
          red: !lambda return 250;
          green: !lambda return 50;
          blue: !lambda return 30;

esp8266:
  board: $board

font:
  - file: _fonts/monobit.ttf
    id: ehmtx_font
    size: 16
    glyphs:  |
      !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@<>/

logger:

binary_sensor:
  - platform: status
    name: "$devicename Status"

# Enable Home Assistant API
api:
  services:
    - service: alarm
      variables:
        icon_name: string
        text: string
      then:
        lambda: |-
          id(rgb8x32)->add_screen(icon_name,text,7,true);
          id(rgb8x32)->force_screen(icon_name);
    - service: screen
      variables:
        icon_name: string
        text: string
      then:
        - ehmtx.add.screen:
            id: rgb8x32
            text: !lambda return text;
            icon_name: !lambda return icon_name;
            alarm: false
    - service: brightness
      variables:
        brightness: int
      then:
        lambda: |-
          id(rgb8x32)->set_brightness(brightness);
    - service: status
      then:
        lambda: |-
          id(rgb8x32)->get_status();
    - service: del_screen
      variables:
        icon_name: string
      then:
        - ehmtx.delete.screen:
            id: rgb8x32
            icon_name: !lambda return icon_name;
    - service: indicator_on
      variables:
        r: int
        g: int
        b: int
      then:
        - ehmtx.indicator.on:
            id: rgb8x32
            red: !lambda return r;
            green: !lambda return g;
            blue: !lambda return b;
    - service: force_screen
      variables:
        icon_name: string
      then:
        - ehmtx.force.screen:
            id: rgb8x32
            icon_name: !lambda return icon_name;
    - service: text_color
      variables:
        r: int
        g: int
        b: int
      then:
        lambda: |-
          id(rgb8x32)->set_text_color(r,g,b);
    - service: clock_color
      variables:
        r: int
        g: int
        b: int
      then:
        - ehmtx.clock.color:
            id: rgb8x32
            red: !lambda return r;
            green: !lambda return g;
            blue: !lambda return b;
    - service: alarm_color
      variables:
        r: int
        g: int
        b: int
      then:
        lambda: |-
          id(rgb8x32)->set_alarm_color(r,g,b);
    - service: indicator_off
      then:
        - ehmtx.indicator.off:
            id: rgb8x32

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $devicename
    password: "InM2TlqVfJe4"

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    variant: WS2812
    pin: $mypin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

number:
  - platform: template
    name: "LED brightness"
    min_value: 0
    max_value: 255
    step: 1
    lambda: |-
      return id(rgb8x32)->get_brightness();
    set_action:
      lambda: |-
        id(rgb8x32)->set_brightness(x);

time:
  - platform: homeassistant
    id: ehmtx_time

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

ehmtx:
  id: rgb8x32
  show_clock: 4
  show_screen: 6
  display8x32: ehmtx_display
  time: ehmtx_time
  duration: 7
  html: true
  font_id: ehmtx_font
  on_next_screen:
    - homeassistant.event:
        event: esphome.next_screen
        data_template:
            iconname: !lambda "return x.c_str();"
            text: !lambda "return y.c_str();"
  icons: 
    - file: _icons/home.gif
      id: thuis
    - file: _icons/thuis-komst.gif
      id: thuis_komst
    - file: _icons/not-home.gif
      id: weg
    - file: _icons/weg-gaan.gif
      id: weg_gaan
    - file: _icons/deur-open.gif
      id: deur_open
    - file: _icons/door-gesloten.gif
      id: deur_gesloten
    - file: _icons/raam-open.gif
      id: raam_open
    - file: _icons/raam-gesloten.gif
      id: raam_gesloten
    - file: _icons/geen-post.png
      id: geen_post
    - file: _icons/wel-post.png
      id: wel_post
    - file: _icons/bewolk.png
      id: bewolkt
    - file: _icons/bliksem.gif
      id: bliksem
    - file: _icons/zon-met-regen.gif
      id: zon_met_regen
    - file: _icons/regen.gif
      id: regen
    - file: _icons/sneeuw-val.gif
      id: sneeuw
    - file: _icons/wind.gif
      id: wind
    - file: _icons/sun.gif
      id: zon
    - file: _icons/maan.png
      id: maan
    - file: _icons/hartje.gif
      id: hartje
    - file: _icons/i-love-you.gif
      id: i_love_you
    - file: _icons/loveman.png
      id: loveman
    - file: _icons/kushartje.png
      id: kushartje
    - file: _icons/i-love.png
      id: i_love
    - file: _icons/restafval.png
      id: restafval
    - file: _icons/pmd.png
      id: pmd
    - file: _icons/papier.png
      id: papier
    - file: _icons/geen-afval.png
      id: geen_afval
    - file: _icons/wel-afval.png
      id: wel_afval
    - file: _icons/Home-Assistant.png
      id: home_assistant
    - file: _icons/HomeAssistantgif.gif
      id: home_assistant_gif
    - file: _icons/instagram.png
      id: instagram
    - file: _icons/kerstboom_sneew.gif
      id: kerstboom_sneew
    - file: _icons/temperature.png
      id: temperature
    - file: _icons/calender.png
      id: agenda
```

lubeda, 2023-02-14:
Hi, can you try this entry als additional icon:

`  
    - url: https://developer.lametric.com/content/apps/icon_thumbs/37472
      id: waschmaschine
`

The Icons from the awtrix icon-database are a little "crazy" formated.

jrspowers, 2023-02-15:
ok i will try it later thanks for the reply

lubeda, 2023-02-17:
I expect a problem with the used icons, if they still don't work perhaps you should scale them to 8x8 gifs with gimp instead off the internal algorithm of ehmtx. => close

closed 2023-02-17 by lubeda