lubeda/EsphoMaTrix #32: leds is not in the right place

bobdivx, 2023-02-14:
Hello, I have a flexible 32x8 led matrix, I sent your scirpt to my esp, but the lighting of the leds is not in the right place, and I can't understand what to modify , and I did not find other examples on the internet, on PixelIT I must set to Row major. Can you help me please; thanks in advance

lubeda, 2023-02-14:
Sorry,
see https://community.home-assistant.io/t/esphomatrix-a-simple-clock-status-display/425325/4?u=lubeda for a possible way to solve your issue.

This problem is more esphome related than [EspHoMaTriX](https://community.home-assistant.io/t/esphomatrix-a-simple-clock-status-display/425325) related.

closed 2023-02-14 by lubeda

lubeda, 2023-02-16:
You can try this pixel mapper, it works with this display 
 [Ulanzi TC1](https://de.aliexpress.com/item/1005005008682055.html)

```
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
```