lubeda/EsphoMaTrix #33: awtrixid bitmaps

bobdivx, 2023-02-17:
Hello I get an error when I want to send the code to my esp32
Error:
`INFO Reading configuration /config/esphome/esphomatrix.yaml...
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
/usr/lib/python3/dist-packages/PIL/Image.py:962: UserWarning: Palette images with Transparency expressed in bytes should be converted to RGBA images
  warnings.warn(
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/requests/models.py", line 971, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 963, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 950, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 395, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 176, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 188, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 619, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 168, in wrapped
    await coro(conf)
  File "/config/esphome/.esphome/external_components/66a08d7e/components/ehmtx/__init__.py", line 438, in to_code
    awtrixdata = r.json()
  File "/usr/local/lib/python3.9/dist-packages/requests/models.py", line 975, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
`

My code:
`display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (y % 2 == 0) {
        return x + ( y * 32);
      }
      else
      {
      return (31 - x) + ( y * 32);
      }
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();
ehmtx:
  ehmtxselect: ehmtx_screens
  id: rgb8x32
  show_clock: 4
  show_screen: 6
  display8x32: ehmtx_display
  time: ehmtx_time
  duration: 60
  font_id: ehmtx_font
  on_next_screen:
    lambda: |-
        ESP_LOGD("Trigger","Ole it works");
  icons: 
    - file: icons/SUN_5807.gif
      id: boot
    - awtrixid: 662
      id: temp
    - file: icons/alert.png
      id: alert
    - awtrixid: 417
      id: door`

lubeda, 2023-02-17:
The awtrix icon database is down! So I have to remove this option from the source and the documentation!

jrspowers, 2023-02-17:
hello if you now ins contact the maker with the maker of awtrix wand who has taken them offline because you just use this he says!

lubeda, 2023-02-17:
As i stated in my documentation **you** have to check the copyright of the icons **you** use! Besides of testing i didn't use his icons!

I removed the direct access to the awtrix icon database.

closed 2023-02-17 by lubeda

jrspowers, 2023-02-17:
yes that is certainly true what you say is just an announcement that you say this & the new update works together with ulanzi TC001 now also work with esphomatrix?

lubeda, 2023-02-17:
I made a customized yaml for the ulanzi TC001 clock. 
The middle button, the battery sensor and the LDR aren't "really" working. 
But the buzzer, rtc, display, temp/humidity sensor are working fine 🎉

The new version of the component was made only for the awtrix bitmaps.

jrspowers, 2023-02-17:
so I can put the new version on the ulanzi TC001 then it will work nicely on that screen?

lubeda, 2023-02-17:
Yes, even the old version. 
It depends all on the yaml. 
My sample yaml has some limitations but the display is working fine

jrspowers, 2023-02-17:
a okay well then how does it work perfzct for that screen which yaml file should I use to make the button and everything work perfectly?