lubeda/EsphoMaTrix #34: ulanzi TC001

jrspowers, 2023-02-18:
hello I just wanted to install your new esphomatrix version for the ulanzi TC001 but I get an error that it does not find it? see code

error:
```yaml
INFO Reading configuration /config/esphome/ulanzi.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Error reading file /config/esphome/ehmtx_font.yaml: [Errno 2] No such file or directory: '/config/esphome/ehmtx_font.yaml'
```

code yaml :
```yaml
substitutions:
  devicename: ulanzi
  ledpin: GPIO32 
  buzzerpin: GPIO15

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EsphoMaTrix
    refresh: 60s 
    components: [ ehmtx ]   

esphome:
  comment: "Ulanzi Work in Progress"
  name: $devicename 
  on_boot:
    then:
      - ds1307.read_time:
  
esp32:
  board: esp32dev

font: !include ehmtx_font.yaml

binary_sensor:
  - platform: status
    name: "$devicename Status"
  - platform: gpio
    pin:
      number: $left_button_pin
      inverted: true
    name: "$devicename left button"
  - platform: gpio
    pin: 
      inverted: true
      number: $mid_button_pin
      mode: INPUT_PULLUP
    name: "$devicename middle button"
  - platform: gpio
    pin: 
      number: $right_button_pin
      inverted: true
    name: "$devicename right button"

logger:
  level: INFO

# Enable Home Assistant API
api:
  services:
    - service: gauge
      variables:
        val: int
      then:
        lambda: |-
          id(rgb8x32).set_gauge_value(val);
    - service: alarm
      variables:
        icon_name: string
        text: string
      then:
        lambda: |-
          id(rgb8x32)->add_screen(icon_name,text,7,true);
          id(rgb8x32)->force_screen(icon_name);
    - service: screen
      variables:
        icon_name: string
        text: string
      then:
        - ehmtx.add.screen:
            id: rgb8x32
            text: !lambda return text;
            icon_name: !lambda return icon_name;
            alarm: false
    - service: brightness
      variables:
        brightness: int
      then:
        lambda: |-
          id(rgb8x32)->set_brightness(brightness);
    - service: status
      then:
        lambda: |-
          id(rgb8x32)->get_status();
    - service: del_screen
      variables:
        icon_name: string
      then:
        - ehmtx.delete.screen:
            id: rgb8x32
            icon_name: !lambda return icon_name;
    - service: indicator_on
      variables:
        r: int
        g: int
        b: int
      then:
        - ehmtx.indicator.on:
            id: rgb8x32
            red: !lambda return r;
            green: !lambda return g;
            blue: !lambda return b;
    - service: force_screen
      variables:
        icon_name: string
      then:
        - ehmtx.force.screen:
            id: rgb8x32
            icon_name: !lambda return icon_name;
    - service: text_color
      variables:
        r: int
        g: int
        b: int
      then:
        lambda: |-
          id(rgb8x32)->set_text_color(r,g,b);
    - service: alarm_color
      variables:
        r: int
        g: int
        b: int
      then:
        lambda: |-
          id(rgb8x32)->set_alarm_color(r,g,b);
    - service: indicator_off
      then:
        - ehmtx.indicator.off:
            id: rgb8x32
  encryption:
    key: "iEvQjMi6GHXTS94Qmda/VlFQZOAo+Qo+T6mvphuaNOU="

number:
  - platform: template
    name: "$devicename brightness"
    min_value: 0
    max_value: 255
    step: 1
    lambda: |-
      return id(rgb8x32)->get_brightness();
    set_action:
      lambda: |-
        id(rgb8x32)->set_brightness(x);

sensor:
  - platform: sht3xd
    temperature:
      name: "$devicename Temperature"
    humidity:
      name: "$devicename Relative Humidity"
    update_interval: 60s
  - platform: adc
    pin: 36
    # not really working
    name: "$devicename Battery"
    update_interval: 60s

  - platform: adc
    id: source_sensor
    # not really working
    name: "$devicename LDR"
    pin: A7

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:

output:
  - platform: ledc
    pin: $buzzerpin
    id: rtttl_out

rtttl:
  output: rtttl_out

i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    variant: WS2812
    pin: $ledpin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

time:
  - platform: homeassistant
    on_time_sync:
      then:
        ds1307.write_time:
  - platform: ds1307
    update_interval: never
    id: ehmtx_time
  
display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

ehmtx:
  id: rgb8x32
  show_clock: 6
  show_screen: 6
  html: true
  display8x32: ehmtx_display
  time: ehmtx_time
  duration: 7
  font_id: ehmtx_font
  icons: !include ehmtx_icons.yaml
```

jrspowers, 2023-02-18:
next errror: 

```yaml
binary_sensor.gpio: [source /config/esphome/esphomatrix-mini.yaml:34]
  platform: gpio
  pin: 
    
    Cannot resolve pin name '$left_button_pin' for board esp32dev.
    number: $left_button_pin
    inverted: True
  name: ulanzi left button
binary_sensor.gpio: [source /config/esphome/esphomatrix-mini.yaml:39]
  platform: gpio
  pin: 
    inverted: True
    
    Cannot resolve pin name '$mid_button_pin' for board esp32dev.
    number: $mid_button_pin
    mode: INPUT_PULLUP
  name: ulanzi middle button
binary_sensor.gpio: [source /config/esphome/esphomatrix-mini.yaml:45]
  platform: gpio
  pin: 
    
    Cannot resolve pin name '$right_button_pin' for board esp32dev.
    number: $right_button_pin
    inverted: True
  name: ulanzi right button
```

lubeda, 2023-02-18:
i fixed the `$right_button_pin` problem. for the font definition you need to check your local path etc. (see the README.md). also the icons, this depends on your configuration!

Please give the new yaml a try.

jrspowers, 2023-02-18:
go out test later thanks for adjusting the version!

jrspowers, 2023-02-18:
hello if I want to flash it with https://web.esphome.io/ then I get a very high pip tone and then it flashes well now all hope if it works!

jrspowers, 2023-02-18:
hello i got him work all few questions
1st question can I change the color of clock/date
2nd I ask if the screen has rebooted itself, the color comes to light pink, how can this be adjusted, the brightness always comes to 80%, I want this at 130%, can this be adjusted, thanks in advance

then I have a comment when sending a message via screen I get a blue bar next to the icon can this be solved?

https://user-images.githubusercontent.com/60328474/219899586-ecfc8f9d-ead4-4e42-b1b3-9c09807b9d1b.mp4

lubeda, 2023-02-19:
All the color settings are described in the README.md. For the weird icon, you should try to use a 8x8 pixel png first, this should work. Then you should try a 8x8 gif this should work as an animation. Can u post your used icon from your sample?

lubeda, 2023-02-19:
The high beep comes from the hardware design of the Ulanzi Clock while the port/pin is uninitialized you hear a high beep sound.

closed 2023-02-20 by lubeda