lubeda/EsphoMaTrix #36: Matrix type

dereik, 2023-02-19:
Hi ,
my 8*32 matrix connected to an esp8266 shows only "random" pixels.
The screen seems to switch between two states, probably date and time, but it is not readable.
When i use awtrix i also had to select matriy type 2 or so. does this option exist here too?

best regards

lubeda, 2023-02-19:
You have to customize the pixel mapper as stated in the esphome manual.

I guess this could work. But it is not part of my code so rtfm.

```yaml
display:
  - platform: addressable_light
    .....
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
    .....
```

closed 2023-02-19 by lubeda

dereik, 2023-02-19:
works now perfekt

thanks