lubeda/EsphoMaTrix #38: Home Assistant and Ulanzi soeed

aptonline, 2023-02-19:


lubeda, 2023-02-19:
The communication between home assistant and the display is asynchronous! 
There is a queue on the display that shows all screens after another, as they came in. See diagram in the README.md. 

If you like to force a screen e.g. "door bell rings" you have to send the screen and  [force](https://github.com/lubeda/EsphoMaTrix#force-screen) it. Even that may take some seconds to show up the screen! Force moves the pointer of the next screen to display to the forced screen. !?! 

This is by design!

aptonline, 2023-02-19:
That’s makes sense now. I was expecting an update as soon as the service was triggered, similar to PixelIt. Thanks for the clarification.

closed 2023-02-19 by aptonline