lubeda/EsphoMaTrix #39: Ulanzi Temperature and Humidity sensors?

aptonline, 2023-02-19:
I noticed when EsphoMaTrix is connected to home assistant that it’s reporting temperature and humidity but from our investigations the Ulanzi doesn’t appear to have such sensors. Is this an omission from the provided sample or have we missed seeing these sensors 🤣.

lubeda, 2023-02-19:
it was guesswork, there is a i2c address 0x44 and I tried all possible i2c chips, that one from the samples worked. The values I get from it could be plausible.  This was trial and error not science!!

wsbtak, 2023-02-19:
That's quite cool! Can you help me to identify the sensors on the board please? So I can update the documentation on my end too 👍

[image]

lubeda, 2023-02-19:
Hi, 
i didn't open my device, i tried some options by software only! I think IC3 is the memory. I can't read whats written on ic1,ic2,ic4,ic5,ic6.

The rtc works fine, the temp/humidity also. There has to be a chip for the charging and the "power" on functionality.  A good battery monitoring would be nice, but i will use this clock only in my office connected to power. You did a very good job in "hacking" the hardware of this device. Nice work

jrspowers, 2023-02-19:
thanks @lubeda & @wsbtak  to all 2 for the beautiful delivery of this software in collaboration with home assistant and esphome & I think it is very good work & battery is not necessary for me specifically wall is also on the power!

wsbtak, 2023-02-21:
Excellent finding @lubeda - there is indeed a super tiny sensor on the back of the PCB next to the buzzer 👍 will update my documentation this week about it. many thanks for tinkering around the sensor 🤗

[image]

wsbtak, 2023-02-21:
> 
Btw, Battery Sensor is GPIO34/ADC6 not 36 . data on the sensor is in RAW format

lubeda, 2023-02-21:
i fixed the batt_pin port in the sample. thanx

lubeda, 2023-02-26:
@wsbtak 
> [image]

Hi, are there any possibilities to connect custom sensors to the board, e.g. another i2c sensor? Is it possible to open the case without destroying it?

pplucky, 2023-02-28:
> > 
> 
> Btw, Battery Sensor is GPIO34/ADC6 not 36 . data on the sensor is in RAW format

Any idea on how to format the obtained values in a human-readable way?

wsbtak, 2023-03-02:
everything above 660 is +/- 100% - did some stress testing and measurement

wsbtak, 2023-03-02:
> @wsbtak
> 
> > [image]
> 
> Hi, are there any possibilities to connect custom sensors to the board, e.g. another i2c sensor? Is it possible to open the case without destroying it?

Not found possibilities without destroying the case and have some extra work to wire/solder additional stuff. You need a hot air gun to open up the case and enough time to open it - otherwise the black acryl glass will crack