lubeda/EsphoMaTrix #42: Clock showing __ __ under, even if only time is set to be displayed

pplucky, 2023-02-28:
Hello.

Just got my Ulanzi clock yesterday and managed to get EsphoMaTrix working on it. Great work, thanks in advance.

In "other" clock applications I used (awtrix, for instance), these __ __ only appeared in case more than 1 information was to be shown (date, day of week, etc).

As per my personal settings, I usually only display the clock, so I wonder if removing the __ __ appearing below the clock is feasible, in case only time is set to be displayed.

Thanks in advance.

andrew-codechimp, 2023-02-28:
According to the docs, should be possible if you add **day_of_week: false** under the ehmtx section though not tried myself.

pplucky, 2023-02-28:
> According to the docs, should be possible if you add **day_of_week: false** under the ehmtx section though not tried myself.

I missed it:
`dayofweek (Optional, bool): draw the day indicator on the bottom of the screen. Disable e.g. if you want larger fonts, defaults to true.`

I'll try it with dayofweek: false, thanks again.,

pplucky, 2023-02-28:
Solved, according to the documentation.

closed 2023-02-28 by pplucky