lubeda/EsphoMaTrix #46: Feature Requests: Allow setting text color as rainbow / set text a number of times

pplucky, 2023-03-04:
Hello.

Recently bought an Ulanzi clock and glad to have EsphoMaTrix working on it.

While using awtrix, there are 2 things which I REALLY miss in EsphoMaTrix (and which I think that fits in its different purpose compared to awtrix), which is to:

- allow to set text color as a rainbow, to allow the color to be set random and changing while the text is being displayed
- allow to set a text to show on display only a specified number of times.

Would this be something potentially implementable in the future?

Thanks in advance.

lubeda, 2023-03-05:
You can get similar results with the [color-actions](https://github.com/lubeda/EsphoMaTrix#set-textalarmclockweekdaytoday-color-action)

So i think, this does not belong to the component, instead you can do it yourself with epshome in your yaml.

pplucky, 2023-03-06:
> You can get similar results with the [color-actions](https://github.com/lubeda/EsphoMaTrix#set-textalarmclockweekdaytoday-color-action)

Fair enough, I'll consider that option.

> * allow to set a text to show on display only a specified number of times.

What about this? It would be useful, wouldn't it?

lubeda, 2023-03-07:
Hi, the "number of times" to display is a problem. Think of only one screen in the queue. If you then specify 2 times its gone after some seconds. So i think the duration is the best way to limit the display time.

For your color idea i tried this:

```
time:
  - platform: homeassistant
    on_time:
      # Every 5 minutes
      - seconds: /1
        then:
         # logger.log: Hallo
         lambda: |-
          id(rgb8x32)->set_text_color(rand() % 255, rand() % 255, rand() % 255);
          id(rgb8x32)->set_clock_color(rand() % 255, rand() % 255, rand() % 255);
```

it changes the color randomly each seconds.

closed 2023-03-07 by lubeda

jrspowers, 2023-03-07:
hello, I just tested the color change, works fine, but it's a pity that he also takes the message in that color with him, is it possible to have this only on the clock?

jrspowers, 2023-03-09:
@lubeda is this possible what i ask above?

lubeda, 2023-03-09:
From the above sample:
this line changes the text color on each tick
`          id(rgb8x32)->set_text_color(rand() % 255, rand() % 255, rand() % 255);`
this line changes the clock color on each tick
`          id(rgb8x32)->set_clock_color(rand() % 255, rand() % 255, rand() % 255);`

in the upcoming 2023.3.4

there are some other possibilities for the clock color:
this changes the clock color for a whole "show_screen" intervall.
```yaml
ehmtx:
  ....
  on_next_clock:
    lambda: |-
      id(rgb8x32)->set_clock_color(rand() % 255, rand() % 255, rand() % 255);
      id(rgb8x32)->set_weekday_color(rand() % 255, rand() % 255, rand() % 255);
      id(rgb8x32)->set_today_color(rand() % 255, rand() % 255, rand() % 255);

For a rainbox like color change you have to create a lambda on your own, but the possibilies are there!

```

jrspowers, 2023-03-09:
hello this works very nicely now i have it in my cofig like this

```yaml
time:
  - platform: homeassistant
    on_time:
      # Every 1 second
      - seconds: /1
        then:
         # logger.log: Hallo
         lambda: |-
          id(rgb8x32)->set_clock_color(rand() % 255, rand() % 255, rand() % 255);
          id(rgb8x32)->set_weekday_color(rand() % 255, rand() % 255, rand() % 255);
          id(rgb8x32)->set_today_color(rand() % 255, rand() % 255, rand() % 255);
```