lubeda/EsphoMaTrix #48: Reference screen by name/id instead of icon

jd1, 2023-03-05:
I would like to propose an API change: I would like to use a screen name or id as a reference to add, update or delete a screen instead of using the name of an icon.

``` c++
void add_screen(std::string screen_name, std::string icon, std::string text, uint16_t duration, bool alarm);
void del_screen(std::string screen_name);
```

This brings two advantages:
* You can change the icon of an existing screen
* You can use the same icon for multiple screens

My main use case is a weather screen: With the current implementation, I have to remove a screen and add a new one everytime when the weather condition changes. If I could call the API with a screen name, the update is much simpler.

I implemented this change I while ago(https://github.com/jd1/EsphoMaTrix/tree/id), but never polished to code to create a PR. I probably has some merge conflicts now :)

For most usecases there is not much that you have to change, but I see one risk: I implemented the name/id of a screen as `std::string` which could consume too much memory. I hope that there is a better solution for this.

On my reference branch I duplicated the `add_screen` and `del_screen` methods to keep the API stable, but I would prefer a clean solution with only two methods.

I am happy to receive feedback on my proposal.

andrew-codechimp, 2023-03-05:
I faced the same requirement and did consider an ID as a breaking change.

To get around the issue I created a wildcard delete to remove any screens with an icon beginning weather_* before sending my latest weather icon.  @lubeda has accepted this for the next release but if more use cases emerge an ID would be the way to go, you could perhaps have the ID as an optional and default to the IconName if missing which would make it easier for casual users/using Notify extension.

Your duplicate add_screen with ID is a good idea.

https://github.com/lubeda/EsphoMaTrix/pull/45

lubeda, 2023-03-05:
I'd never thought about this use case, since Andrews " weather_*" idea i implemented a weather and a trash routine for me. Thanx Andrew.

I have to think about this request a little bit.

lubeda, 2023-03-13:
I understand the requirement and see the possible uses, but it is not necessary for my purposes. An implementation would require a redesign of the inner processes.
If someone has an implementation, I am happy about every pull request.

That's why this topic is tagged with "help wanted"

andrewjswan, 2023-10-12:
@jd1 look https://github.com/lubeda/EspHoMaTriXv2/issues/83

lubeda, 2023-10-31:
available in esphomatrixv2

closed 2023-10-31 by lubeda