lubeda/EsphoMaTrix #49: Feature Request: Update screen immediately on force screen

jd1, 2023-03-05:
It would be nice if the screen would update immediately when you call force screen.
E.g. I would like to show the current caller on my landline phone and it is annoying if I have to wait until the screen moves to the next to see the caller.

If you would like to keep the current behaviour, you could also split it up in two methods:
* Set next screen: This is the current behaviour of _force screen_
* Force screen: This updates the current screen immediately

It's been over half a year since I had a deeper look at the code, so I'm not sure how much effort this is. This is just an idea for an improvement, no show stopper :)

lubeda, 2023-03-07:
Hi, take a look at the branch 2023.3.3, this should fix it!

jd1, 2023-03-07:
That's a nice fix 👍
I didn't test it yet, but the code looks good.

closed 2023-03-09 by lubeda

jrspowers, 2023-03-09:
hello @lubeda  if add this to the config for ulanzi it stays red it is still missing something in the config of the hold screen but can't find the service setting anywhere?

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: $right_button_pin
      inverted: true
    on_press:
      lambda:
        id(rgb8x32)->hold_screen();
```

jrspowers, 2023-03-09:
i never have to do anything when you release update work all good of new things ?  so hence this question?

jrspowers, 2023-03-09:
@lubeda i know what the problem was forgetting to put the name in & what does the hold_screen mean to stay on that screen for 20 sec?

jrspowers, 2023-03-30:
> to only show e.g. the youtubesubscribers, i recommend:
> 
> disable the clockdisplay
> 
> ```
> 
> esphome:
> 
>   on_boot:
> 
>     then:
> 
>       - ds1307.read_time:
> 
>       - lambda:
> 
>          id(rgb8x32)->set_clock_time(0);
> 
> ```
> 
> read the value from home assistant
> 
> 
> 
> ```
> 
> sensor:
> 
>   - platform: homeassistant
> 
>     name: "Github sensor from home assistant"
> 
>     id: sgithub
> 
>     entity_id: sensor.lubeda_esphomatrix_stars
> 
>     on_value:
> 
>       lambda: 
> 
>         ESP_LOGI("github", "Value is %f", id(sgithub)->state);
> 
> ```
> 
> 
> 
> refresh the display every minute
> 
> 
> 
> ```
> 
> time:
> 
>   - platform: ds1307
> 
>     id: ehmtx_time
> 
>     on_time:
> 
>       - seconds: 0
> 
>         minutes: /1
> 
>         then: 
> 
>           lambda: |-
> 
>             char text[20];
> 
>             sprintf(text,"%10.0f", id(sgithub).state);
> 
>             id(rgb8x32)->add_screen("github", text, 2,61, false);      
> 
> ```
> 
> 
> 
> Only the subscriber count comes from homeassistant the rest is only in the esphome yaml.

yes okay good but if I now want it to work again I want to switch it on with the middle button, for example 1 click & 2 clicks to turn it off, is that possible?  in this way ?