lubeda/EsphoMaTrix #55: screen brightness

jrspowers, 2023-03-09:
hello is it possible to set the startup screen brightness to your own set brightness ? so that it doesn't have to be done every time?

pplucky, 2023-03-09:
Have you tried this?

```
esphome:
  ....
  on_boot:
    priority: -100
    # ...
    then:
      - lambda: !lambda |-
          id(rgb8x32)->set_brightness(YOURVALUE);
```

I think this should force the brightness to be set to YOURVALUE on esp boot (similar to [show_all_icons example in documentation](https://github.com/lubeda/EsphoMaTrix#show-all-icons-on-your-matrix)).

jrspowers, 2023-03-09:
thank you I will try that later thank you 😊

pplucky, 2023-03-09:
> thank @pplucky you this work top & I have another question after restart my screen is always off on my esphomatrix screen can't find where this setting is on my wall ulanzi isn't this the case?

You mean when you restart/boot the esp32, right? 

I only restart mine when I flash it, otherwise I just turn the display off and on with the [corresponding service or switch](https://github.com/lubeda/EsphoMaTrix#services).

Still then, when I restart the esp32, it feels like it doesn't always behave the same way, but it also happens to me some times (not all the time).

What I've seen also - not sure if that's your case, when I'm playing with esphome light effects, if the effect ends up in full black and I don't finish the effect (with effect 'None') or turn off the light, matrix stays all black and doesn't show anything EsphoMaTrix-related (clock or screens).

jrspowers, 2023-03-09:
yes I mean so if you have updated the config via esphome this sava and install wireless then it restarts anyway & then the screen is black I don't understand why?

pplucky, 2023-03-09:
You can always try:

esphome:
  ....
  on_boot:
    priority: -100
    # ...
    then:
      - lambda: !lambda |-
          id(rgb8x32)->set_display_on();

But this would only be a workaround anyway. 

Not sure if someone else can try to help a bit more on this (for me), non-consistent behavior..

jrspowers, 2023-03-09:
thank you very much for helping me this work only at perfect !

closed 2023-03-09 by lubeda