lubeda/EsphoMaTrix #56: Question: How to better include turn off animation

pplucky, 2023-03-09:
In my awtrix matrix, there is a turn off animation whenever the display is turned off, which I think it is nice.

I built an [Addressable Lambda Effect](https://esphome.io/components/light/index.html#addressable-lambda-effect) in esphome to mimic the same animation on my Ulanzi EsphoMaTrix, assigned it to the light component and it works fine if I trigger the effect manually.

Probably there are simpler ways to achieve this, but here it is for reference, if anyone wants to use it also:
```
addressable_lambda:
  name: "ZZZ"
  update_interval: 100ms
  lambda: |-
    static int x = 0;
    if (initial_run) {
      x = 1;
      it.all() = ESPColor(0,0,0);
    }
    // it.range() seems to include initial LED # but excludes last LED #
    switch (x) {
      case 1:
        // Line 8 - right to left (from 224)
        it.range(231, 236) = ESPColor(255,255,255);
        x += 1;
        break;
      case 2:
        // Line 7 - left to right (from 192)
        it.range(210, 215) = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it.range(231, 236) = ESPColor(0,0,0);
        it[233] = ESPColor(255,255,255);
        x += 1;
        break;
      case 3:
        // Line 6 - right to left (from 160)
        it.range(172, 177) = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it.range(210, 215) = ESPColor(0,0,0);
        it[211] = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it[233] = ESPColor(0,0,0);
        it[237] = ESPColor(255,255,255);
        x += 1;
        break;
      case 4:
        // Line 5 - left to right (from 128)
        it.range(141, 146) = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it.range(172, 177) = ESPColor(0,0,0);
        it[174] = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it[211] = ESPColor(0,0,0);
        it[208] = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it[237] = ESPColor(0,0,0);
        it[240] = ESPColor(255,255,255);
        x += 1;
        break;
      case 5:
        // Line 4 - right to left (from 96)
        it.range(111, 116) = ESPColor(255,255,255);
        // Line 5 - left to right (from 128)
        it.range(141, 146) = ESPColor(0,0,0);
        it[144] = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it[174] = ESPColor(0,0,0);
        it[176] = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it[208] = ESPColor(0,0,0);
        it[206] = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it[240] = ESPColor(0,0,0);
        it[242] = ESPColor(255,255,255);
        x += 1;
        break;
      case 6:
        // Line 3 - left to right (from 64)
        it.range(75, 80) = ESPColor(255,255,255);
        // Line 4 - right to left (from 96)
        it.range(111, 116) = ESPColor(0,0,0);
        it[112] = ESPColor(255,255,255);
        // Line 5 - left to right (from 128)
        it[144] = ESPColor(0,0,0);
        it[142] = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it[176] = ESPColor(0,0,0);
        it[178] = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it[206] = ESPColor(0,0,0);
        it[204] = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it[242] = ESPColor(0,0,0);
        it[244] = ESPColor(255,255,255);
        x += 1;
        break;
      case 7:
        // Line 2 - right to left (from 32)
        it.range(49, 54) = ESPColor(255,255,255);
        // Line 3 - left to right (from 64)
        it.range(75, 80) = ESPColor(0,0,0);
        it[78] = ESPColor(255,255,255);        
        // Line 4 - right to left (from 96)
        it[112] = ESPColor(0,0,0);
        it[114] = ESPColor(255,255,255);
        // Line 5 - left to right (from 128)
        it[142] = ESPColor(0,0,0);
        it[140] = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it[178] = ESPColor(0,0,0);
        it[180] = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it[204] = ESPColor(0,0,0);
        it[202] = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it[244] = ESPColor(0,0,0);
        it.range(241, 246) = ESPColor(255,255,255);
        x += 1;
        break;
      case 8:
        // Line 1 - left to right (from 0)
        it.range(8, 13) = ESPColor(255,255,255);
        // Line 2 - right to left (from 32)
        it.range(49, 54) = ESPColor(0,0,0);
        it[51] = ESPColor(255,255,255);
        // Line 3 - left to right (from 64)
        it[78] = ESPColor(0,0,0);
        it[75] = ESPColor(255,255,255);
        // Line 4 - right to left (from 96)
        it[114] = ESPColor(0,0,0);
        it[117] = ESPColor(255,255,255);
        // Line 5 - left to right (from 128)
        it[140] = ESPColor(0,0,0);
        it[137] = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it[180] = ESPColor(0,0,0);
        it[183] = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it[202] = ESPColor(0,0,0);
        it.range(200, 206) = ESPColor(255,255,255);
        // Line 8 - right to left (from 224)
        it.range(241, 246) = ESPColor(0,0,0);
        x += 1;
        break;
      case 9:
        // Line 1 - left to right (from 0)
        it.range(8, 13) = ESPColor(0,0,0);
        it[13] = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it[183] = ESPColor(0,0,0);
        it.range(178, 183) = ESPColor(255,255,255);
        // Line 7 - left to right (from 192)
        it.range(200, 206) = ESPColor(0,0,0);
        x += 1;
        break;
      case 10:
        // Line 5 - left to right (from 128)
        it[137] = ESPColor(0,0,0);
        it.range(138, 143) = ESPColor(255,255,255);
        // Line 6 - right to left (from 160)
        it.range(178, 183) = ESPColor(0,0,0);
        x += 1;
        break;
      case 11:
        // Line 4 - right to left (from 96)
        it[117] = ESPColor(0,0,0);
        it.range(112, 117) = ESPColor(255,255,255);
        // Line 5 - left to right (from 128)
        it.range(138, 143) = ESPColor(0,0,0);
        x += 1;
        break;
      case 12:
        // Line 3 - left to right (from 64)
        it[75] = ESPColor(0,0,0);
        it.range(76, 81) = ESPColor(255,255,255);
        // Line 4 - right to left (from 96)
        it.range(112, 117) = ESPColor(0,0,0);
        x += 1;
        break;
      case 13:
        // Line 2 - right to left (from 32)
        it[51] = ESPColor(0,0,0);
        it.range(46, 51) = ESPColor(255,255,255);
        // Line 3 - left to right (from 64)
        it.range(76, 81) = ESPColor(0,0,0);
        x += 1;
        break;
      case 14:
        // Line 1 - left to right (from 0)
        it[13] = ESPColor(0,0,0);
        it.range(14, 19) = ESPColor(255,255,255);
        // Line 2 - right to left (from 32)
        it.range(46, 51) = ESPColor(0,0,0);
        x += 1;
        break;
      case 15:
        it.all() = ESPColor(0,0,0);
        break;
    }
```

If I turn on the light with the effect, it switches from the clock to the animation. 
The "problem" is that to stop the animation, I either need: 
- to apply an effect 'None' (which actually turns on all the LEDs in the matrix, same as turning light on without any effect) or;
- directly turn off the light (which actually ["reactivates the clock"](https://github.com/lubeda/EsphoMaTrix/tree/main#use-the-light-component) - which makes sense but not when turning off the display).

So my question here is: how would you guys better include this to behave as a turn off animation? Going from clock or whatever is on the display to the animation and then display off.

Thanks for any suggestions.

lubeda, 2023-03-09:
After a bit of thinking....

Perhaps it is best to not use an effect, instead add your code to the display lambda. But this is called every 16ms, so you have to slow down your code and disable my tick() and my draw() functions.

Even better support for 8x32 gifs.....

pplucky, 2023-03-10:
> i think the idea with 8x32 animations would be possible with out changing the whole code.
> like 8x8 icons with scrolling text and 8x32 animations without text
> i will take a closer look.

I didn’t want to suggest that, considering the amount of priceless work you have already put into this component (thanks again for that), but that would be a perfect addition which I think a lot will appreciate.

lubeda, 2023-03-11:
i added support for 8x32 images and animations. Their display works the same as for 8x8 icons but for 8x32 images no text will be displayed.
breaking: you have to scale all icons your self, i removed the automatic scaling

pplucky, 2023-03-12:
> i added support for 8x32 images and animations (in the 8x32 branch). Their display works the same as for 8x8 icons but for 8x32 images no text will be displayed.

Quick test seems to work, need more time into it.

> breaking: you have to scale all icons your self, i removed the automatic scaling

Noticed the error while compiling before installing to esphome, I just wonder if it would be possible to show the whole list of non 8x8 or 8x32 images, instead of the first found from the list, to make it easier to fix without going one by one.

lubeda, 2023-03-12:
check the recent push to 8x32. icons with wrong size are skipped with warning!

pplucky, 2023-03-13:
For me, it is rather cumbersome to NOT to be able to specify how many times a screen is to be shown, even more with animations...

I know you have mentioned it [here](https://github.com/lubeda/EsphoMaTrix/issues/46#issuecomment-1458663539), but  I would still see it as useful, specially with animations.

To be able to achieve an animation to be shown only once (at shutdown, as per my initial question), I'd need to make sure I would delete the screen before the animation would start a second time...

lubeda, 2023-03-13:
I understand the requirement and see the possible uses, but it is not necessary for my purposes. An implementation would require a redesign of the inner processes. 
If someone has an implementation, I am happy about every pull request.

That's why this topic is tagged with "help wanted"

ofirsnb, 2023-03-13:
> Quick test seems to work, need more time into it.

@pplucky 
Can you share a 8x32 animation gif for testing please?

pplucky, 2023-03-13:
> @pplucky Can you share a 8x32 animation gif for testing please?

Sure, here is the 8x32 GIF I generated for the above effect:
[image]

lubeda, 2023-03-26:
starting with 2023.4.0 there is a parameter show_time for each screen, you can set a longer displaytime per screen.
Please read the "how to update" section in the new README.md before trying!

If you included your sample gif under the id `sleeping` you can display it for e.g. 60 seconds with this call:

```
service: esphome.ulanzi_add_screen
data:
  icon_name: sleeping
  text: ehmtx
  screen_time: 60
```

With the possibility to disable the clock you can show this animation all day long.

closed 2023-03-26 by lubeda