lubeda/EsphoMaTrix #61: Force screen doesn't reset the timer

cacordina, 2023-03-16:
Hi,

When I use the force screen feature, the timer is not reset. Depending on when the service is called, the message can sometimes be barely visible. Would it be possible to force the reset of the timer? 

Thanks in advance.

lubeda, 2023-03-17:
Which timer? The lifetime in minutes, e.g. if a screen has a lifetime of 5 minutes and you force a screen after 4 minutes it will only be displayed 1 minute longer?
Or another timer?

cacordina, 2023-03-17:
Hi, I was not talking about the lifetime but ( if i am not wrong ) the next_action_time.

I use the force_screen to display a notification instantly but depending on the moment called, it happens that the message does not have time to be displayed completely and the next screen is directly shown.

Looking at the code a bit more, I found a method "hold_screen", which does what I want:
<code>
id(rgb8x32)->add_screen(icon_name,text,2,false);
id(rgb8x32)->force_screen(icon_name);
id(rgb8x32)->hold_screen();
</code>

Also do we have a way to force a screen even if we are showing the time. It seems that if I call force_screen and time is actually show, the screen will be display after that time period is finished.

lubeda, 2023-03-21:
Well i adapted your idea, but i dont add screen_time. Instead i just start a new lifetime with the force.. Its in the 8x32 branch wich will be released soon.

closed 2023-03-21 by lubeda