lubeda/EsphoMaTrix #67: Text not displayed completely

dg1yds, 2023-03-23:
Hello Ludger,

I have the following setup of screens.

```
[19:53:27][I][EHMTX:287]: status time: 23.3.2023 19:53
[19:53:27][I][EHMTX:290]: status brightness: 80 (0..255)
[19:53:27][I][EHMTX:291]: status default duration: 10
[19:53:27][I][EHMTX:292]: status date format: %d.%m.
[19:53:27][I][EHMTX:293]: status time format: %H:%M
[19:53:27][I][EHMTX:294]: status text_color: RGB(40,128,248)
[19:53:27][I][EHMTX:295]: status alarm_color: RGB(248,48,24)
[19:53:27][I][EHMTX:302]: status indicator off
[19:53:27][I][EHMTX:306]: status display on
[19:53:27][I][EHMTX:141]: status active slot: 8
[19:53:27][I][EHMTX:142]: status screen count: 12 of 24
[19:53:27][I][EHMTX:149]: status slot 0 icon 22 text: 19:53 alarm: 0 dd: 8 sec end: 573 sec
[19:53:27][I][EHMTX:149]: status slot 1 icon 3 text: WOHNZIMMER alarm: 0 dd: 14 sec end: 393 sec
[19:53:27][I][EHMTX:149]: status slot 2 icon 0 text: 22.2 alarm: 0 dd: 7 sec end: 398 sec
[19:53:27][I][EHMTX:149]: status slot 3 icon 2 text: 50% alarm: 0 dd: 7 sec end: 403 sec
[19:53:27][I][EHMTX:149]: status slot 4 icon 1 text: 1088 alarm: 0 dd: 8 sec end: 408 sec
[19:53:27][I][EHMTX:149]: status slot 5 icon 7 text: SCHLAFZIMMER alarm: 0 dd: 16 sec end: 408 sec
[19:53:27][I][EHMTX:149]: status slot 6 icon 4 text: 20.7 alarm: 0 dd: 7 sec end: 413 sec
[19:53:27][I][EHMTX:149]: status slot 7 icon 6 text: 53% alarm: 0 dd: 7 sec end: 418 sec
[19:53:27][I][EHMTX:149]: status slot 8 icon 5 text: 970 alarm: 0 dd: 7 sec end: 423 sec
[19:53:27][I][EHMTX:149]: status slot 9 icon 17 text: DRAUSSEN alarm: 0 dd: 12 sec end: 423 sec
[19:53:27][I][EHMTX:149]: status slot 10 icon 16 text: 13.2 alarm: 0 dd: 7 sec end: 428 sec
[19:53:27][I][EHMTX:149]: status slot 11 icon 23 text: Dies ist ein langer Text. 12345678910111213141516171819202122232425 alarm: 0 dd: 12 sec end: 548 sec
```

The last screen shall display a longer text. It is send via the HA screen service.
But it is not shown completely - the first loop is interrupted at "Dies ist ein lan", the 2nd repetition is interrupted at "Dies ist ein".

It did not check if the issue occurs only in combination with the other screens, or if this is a general topic.

I expect somehow a limitation of the text length, but this seems to be an unexpected behavior.

lubeda, 2023-03-25:
I see the effect, but have no clue yet

lubeda, 2023-03-26:
I will investigare and fix it in the upcoming version 2023.4.0

lubeda, 2023-03-29:
fixed in 2023.4.0, it was the type of the variables

closed 2023-03-29 by lubeda