lubeda/EsphoMaTrix #71: Allow to set distinct text_color per screen

pplucky, 2023-03-27:
Seeing `show_time` as new parameter for `add_screen` service starting from 2023.4.0, has reminded of an issue/limitation that might be solved in a similar way.

Allowing a `text_color` to be set differently for each screen, rather than using the last set `text_color` for each active screen. The `text_color` as of today could be used as a global screen default, optionally changeable per screen and valid for that screen only.

Right now if I have an active screen with a color and I start a new simultaneous screen with a different color, both screens will keep the new color.

Would this be something possibly/easily implementable?

Thanks in advance for all the great work you've been doing in this custom component.

lubeda, 2023-03-28:
Hi it's not well tested but in the 2023.4.0 there is a set_screen_color("iconname",rig,b) service but yet no action. I will work on it at weekend

pplucky, 2023-03-28:
> Hi it's not well tested but in the 2023.4.0 there is a set_screen_color("iconname",rig,b) service but yet no action. I will work on it at weekend

OK, that would allow me to call the new `set_screen_color` rather than the existing `set_text_color`. Good luck and once again thanks for the great work.

lubeda, 2023-03-29:
the  optional `set_screen_color` is only for the actual screen for the specified icon. the set_text_color sets the default color!

pplucky, 2023-04-02:
> the optional `set_screen_color` is only for the actual screen for the specified icon. the set_text_color sets the default color!

Just installed brand new [2023.4.0](https://github.com/lubeda/EsphoMaTrix/releases/tag/2023.4.0) and new service `set_screen_color` doesn't seem to have an effect on the screen text color.

I called the service with the following data:
```
  data:
    icon_name: ha_logo
    r: 107
    g: 174
    b: 93
```

and I saw the following in esphome debug log:
```
[00:24:58][D][EHMTX:135]: icon: ha_logo found id: 0
[00:24:58][D][EHMTX:041]: set screen color: icon 0 r: 107 g: 174 b: 93
[00:24:58][D][EHMTX:135]: icon: ha_logo found id: 0
[00:24:58][D][EHMTX:016]: findfreeslot for icon: 0
[00:24:58][D][EHMTX:136]: display length text: Hello!!! pixels 29 calculated: 10 show_time: 10 default: 10
[00:24:58][D][EHMTX:401]: add_screen icon: 0 iconname: ha_logo text: Hello!!! lifetime: 1 screen_time: 10 alarm: 0
[00:25:04][D][EHMTX:289]: next action: show screen "ha_logo" for 10 sec
```

but the text color shown on the matrix is all white...

Any ideas?

Thanks in advance.

pplucky, 2023-04-03:
> So it's best to set the screen color in your automation after every add_screen.

I was setting the color before adding the screen, so it was always picking up the default color (as the screen was not yet in the queue). All working perfectly after changing the order of the service calls in my HA script.

Don't know what you think, but maybe it's worth mentioning this detail in the documentation.