lubeda/EsphoMaTrix #80: Any possible way for showing a timer

AlexanderGlogger, 2023-04-03:
I was wondering if it is possible to display a timer using EsphoMaTrix. E.g. a service is called in Home Assistant, which leads to an timer of specified duration (e.g. 5 minutes) being displayed on the matrix display.
Or are there any workarounds that would enable such functionality.

P.S. Thank you for developing this project.

AlexanderGlogger, 2023-04-03:
Thanks for the help and pointing me in the right direction. To clarify, I create a timer in an automation which then sends the new text to the display every second, right?

lubeda, 2023-04-03:
yes, that should do the trick.

lubeda, 2023-04-09:
Hi,
i found a way. If you have an timer helper in home assistant e.g. `timer.eier_timer` you can uses this automation:

```
alias: Display Timer on ulanzi
description: ""
trigger:
  - platform: state
    entity_id:
      - timer.eier_timer
    to: active
condition:
  - condition: state
    entity_id: timer.eier_timer
    state: active
action:
  - service: esphome.ulanzi_add_screen
    data:
      alarm: false
      icon_name: timer
      screen_time: 10
      lifetime: 1
      text: >-
        {{ (state_attr('timer.eier_timer', 'finishes_at') | as_datetime -
        now()).total_seconds() | int(0) }}
  - service: esphome.ulanzi_force_icon_screen
    data:
      icon_name: timer
  - delay:
      hours: 0
      minutes: 0
      seconds: 1
      milliseconds: 0
  - service: automation.trigger
    data:
      skip_condition: false
    target:
      entity_id: automation.display_timer_on_ulanzi
mode: restart
```

You can optimize a lot of things in this automation but it works.

If you like my work, please consider giving me a star on github.