lubeda/EsphoMaTrix #87: Feature Request - Custom Fonts per screen

scarolan, 2023-04-07:
Some screens benefit from a tiny font. For example when you want to display a 5-digit number.

Other screens only need 3-4 digits (temperature, humidity, etc) and look nice with a larger font.

This FR is to allow an optional `screen_font:` that would override the default font.

lubeda, 2023-04-07:
Well, i thought about this but first only for the clock. For the clock you need a font with numbers at 6 pixels high. For normal screen 7 to 8 pixels are also fine.

I will consider this in an upcoming release!

scarolan, 2023-04-07:
I think the x/y offset setting should also be configurable or maybe even a subkey of the font.  Thanks for your hard work on this, the results are amazing, so easy to configure from home assistant.

lubeda, 2023-04-09:
in the upcoming 2023.5.0 will be two fonts e.g. one with 6 pixel height (for date and time) and a font 8 pixel heigh for the rest. its working in the alpha version.

closed 2023-04-09 by lubeda

scarolan, 2023-06-01:
@lubeda you are the real MVP.  <3 my esphomatrix devices