lubeda/EsphoMaTrix #88: Incorrect timezone

TheGroundZero, 2023-04-07:
I flashed my Ulanzi TC001 with ESPHome using the YAML from [Blakadder's blog post](https://blakadder.com/esphome-pixel-clock/), which seems to resemble the template in this repo.

However, it appears the time on the clock is UTC while I'm currently in UTC+2 (CEST).

My HA is set to the correct timezone.

How can I fix the time on my Ulanzi?

lubeda, 2023-04-07:
Well if you like my work: donated me a star on github.

The solution is RTFM: https://esphome.io/components/time/index.html

esphome tries to detect your timezone if it doesn't work you have to specify it.

TheGroundZero, 2023-04-07:
> esphome tries to detect your timezone, if that doesn't work you have to specify it manualy.

I haven't encountered issues with that in the past.
But maybe I just got unlucky this time. Will manually set the timezone.

closed 2023-04-07 by TheGroundZero