lubeda/EsphoMaTrix #90: Schrift nach oben verschoben

frittes, 2023-04-09:
Hallo,

leider ist bei meiner Matrix die Schrift nach oben hin verschoben. Das Icon sowie die Balken für die Wochentage werden an der richtigen Position angezeigt. Auch wenn ich die Light-Entität aktiviere leuchten alle LEDs in weiß.

Ich habe die Standard yaml für D1 Mini mit dem entsprechenden Pixel-Mapper für die flexible Matrix verwendet.

Ist das ein Konfigurationsfehler von meiner Seite?

[image]

lubeda, 2023-04-09:
Moin, abhängig von der Schriftart kannst Du die Schrift mit  `yoffset: ` nach unten schieben, eventuell musst du die Schrift auch noch verkleinern mit dem `size` parameter unter der `font` section.

Nebenbei, ich freue mich über jeden Stern auf Github und über jeden Sponsor.

frittes, 2023-04-09:
Funktioniert, Danke!

closed 2023-04-09 by frittes