lubeda/EsphoMaTrix #95: Fresh install off sample yaml for Ulanzi fails

Jeltel, 2023-04-16:
I'm trying to run EsphoMaTrix on the Ulanzi. Triggerd by the post of Blackadder off course.

However I'm not able to run EsphoMaTrix on the Ulanzi. I can flash a default ESPhome installation on the Ulanzi, as well as Awtrix Light.

I get the following error: 
```
[11:06:11]ELF file SHA256: 0000000000000000
[11:06:11]
[11:06:11]Rebooting...
[11:06:12][I][logger:258]: Log initialized
[11:06:12][I][EHMTX:494]: set_brightness 80 => 31.37 %
[11:06:12][I][EHMTX:480]: weekstart: monday
[11:06:12][I][EHMTX:467]: show day of week
[11:06:12][I][EHMTX:441]: show date
[11:06:12][I][EHMTX:458]: don't show seconds
[11:06:12][I][app:029]: Running through setup()...
[11:06:12][I][i2c.arduino:175]: Performing I2C bus recovery
[11:06:12][E][i2c.arduino:191]: Recovery failed: SCL is held LOW on the I2C bus
[11:06:17]E (10434) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[11:06:17]E (10434) task_wdt:  - loopTask (CPU 1)
[11:06:17]E (10434) task_wdt: Tasks currently running:
[11:06:17]E (10434) task_wdt: CPU 0: IDLE
[11:06:17]E (10434) task_wdt: CPU 1: IDLE
[11:06:17]E (10434) task_wdt: Aborting.
[11:06:17]
[11:06:17]abort() was called at PC 0x40104c74 on core 0
[11:06:17]
[logs_ulanzi-desktop_run.txt](https://github.com/lubeda/EsphoMaTrix/files/11241743/logs_ulanzi-desktop_run.txt)

[11:06:17]
[11:06:17]Backtrace:0x40083da9:0x3ffbea9c |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40083da9: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
```
This repeats a couple of times.

Full log attached.

Is there something wrong with the configuration?

I'm using the shortest UBS-C cable I have (about 15cm) directly on the server running ESPhome.
[logs_ulanzi-desktop_run.txt](https://github.com/lubeda/EsphoMaTrix/files/11241751/logs_ulanzi-desktop_run.txt)

lubeda, 2023-04-16:
can you provide your configuration? Especially the configuration from blakadder should work fine.

Jeltel, 2023-04-17:
Sure.

Only changed some icons and removed the API password. Otherwise it's the same as the example in this git.

[ulanzi-desktop.yaml.txt](https://github.com/lubeda/EsphoMaTrix/files/11247575/ulanzi-desktop.yaml.txt)

lubeda, 2023-04-17:
Well, your yaml is clean, i see no problems there. Do you have other working esphome devices running with your installation?

`panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402` looks like a local problem in your esphome installation.

Jeltel, 2023-04-17:
Yes I have two ESP powered devices. They've been running for some time now. 

It's the default esphome installation within Home Assistant. 

And i'm not using a user ficeto. Good spot. Looks like something local crept into the esphome release then?

lubeda, 2023-05-14:
Hi, the [new](https://github.com/lubeda/EsphoMaTrixv2/) version has a working yaml for the ulanzi. I will not fix this in this version!

closed 2023-05-14 by lubeda