lubeda/EsphoMaTrix #97: Show more indicators on clock screen

RikMuilwijk, 2023-04-19:
Is it possible to light up a single led on the clocks screen, like alarm or indicator but smaller?

I’d like to add two small indicators next to each other on the clock screen. Or maybe I could modify the source code to change the size and location of the current indicator service?

RikMuilwijk, 2023-04-21:
Yes, I would like to use the indicator or alarm indicator. So how do I define size? From the docs I can only see how to pass a color parameter.

Also, I would like to have one more indicator on the clock screen. But maybe I have to modify code abit?

lubeda, 2023-04-22:
Where would you position the third indicator?

RikMuilwijk, 2023-04-22:
I would place an addressable pixels in the upper left corner, skip a pixel to the right, and then place a second addressable pixel. These two pixels would indicate my corresponding two alarms in home assistant (green for set, red for not set)

the third indicator would than be the already existing indicator in the right corner. I would use that one to indicate that the window is open/closed. The alarm indicator is a bit too big for me, considering the day of week is directly next to it on a Friday.

I don’t know if this makes sense, I really like the work on this project already done btw! It’s kind of awesome to show any given info that is available in home assistant to this nice retro display

lubeda, 2023-06-03:
So, in the [version](https://github.com/lubeda/EsphoMaTrixv2) there are three indicators, alarm, lindicator and rindicator. They have got a different priority, e.g., the lindicator (left indicator) is not displayed with icons because this would "ruin" the icon.

I will not add features to this version anymore!!

closed 2023-06-03 by lubeda