lubeda/EsphoMaTrix #98: One screen keeps repeating, skipping the clock

nukefrenzy, 2023-04-21:
Here's what is going on:
1)  Only the clock is showing.  No other screens rotating through.
2)  HA adds a single screen showing an alert:
```
service: esphome.ulanzi_add_screen
data:
  alarm: false
  icon_name: doorbell
  lifetime: 1
  screen_time: 10
  text: Doorbell
```
The major settings from my ehmtx section are:
```
clock_time: 5            # duration to display the clock after this time the date is displayed until next "show_screen"
clock_interval: 60       # show the clock at least each x seconds (default=60)
screen_time: 10          # duration to display a screen or a clock/date sequence (default=10)
scroll_count: 2          # minimum number of times to scroll large text
```
3)  However, the alert screen plays many times before the clock comes back.  Every 10 seconds I can see it reset and start over.  Eventually the clock comes back.

It's my understanding that it should rotate: clock (10s)->doorbell screen (10s)->clock (10s)->doorbell screen (10s)->etc. until the 1 minute life of the doorbell screen expires.  Then back to only the clock.  Is my understanding correct?  This is not what's happening.  Instead, it just keeps showing the doorbell screen over and over until it expires.

Here's my entire setup for reference:
```
substitutions:
  devicename: ulanzi
  matrix_pin: GPIO32
  buzzer_pin: GPIO15
  left_button_pin: GPIO26
  mid_button_pin: GPIO27
  right_button_pin: GPIO14
  ldr_pin: GPIO35

# Github source files for EsphoMaTrix
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EsphoMaTrix
    components: [ ehmtx ]

esphome:
  comment: Ulanzi TC001 LED Clock
  name: $devicename 
  friendly_name: Clock
  on_boot:
    then:
      - ds1307.read_time:

esp32:
  board: esp32dev

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Fonts and characters
font: 
  - file: Calcium.ttf
    id: ehmtx_font
    size: 16
    glyphs:  |
      !?"%‰()+*=,-_.:°µ²³0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz€$@<>/

# EsphoMaTrix Display Interface: https://github.com/lubeda/EsphoMaTrix
ehmtx:
  id: rgb8x32
  time_component: ehmtx_time
  matrix_component: ehmtx_display
  clock_time: 5            # duration to display the clock after this time the date is displayed until next "show_screen"
  clock_interval: 60       # show the clock at least each x seconds (default=60)
  screen_time: 10          # duration to display a screen or a clock/date sequence (default=10)
  scroll_count: 2          # minimum number of times to scroll large text
  date_format: "%m/%d"     # defaults "%d.%m" (use "%m.%d" for the US)
  time_format: "%I:%M"     # defaults "%H:%M" (use "%I:%M%p" for the US)
  show_dow: false          # draw the day indicator on the bottom of the screen, defaults to true
  show_date: false         # show the date for show_screen - show_clock seconds otherwise only shows the clock for show_screen seconds, defaults to true
  week_start_monday: false # default monday is first day of week, false = sunday
  yoffset: 8               # the text is aligned BASELINE_LEFT, the baseline defaults to 6
  xoffset: 1               # the text is aligned BASELINE_LEFT, the left defaults to 1
  scroll_interval: 80      # the interval in ms to scroll the text (default=80), should be a multiple of the update_interval from the display (default= 16ms)
  frame_interval: 192      # the interval in ms to display the next anim frame (default=192), should be a multiple of the update_interval from the display (default= 16ms)
  font_id: ehmtx_font
  icons: 
    - id: bitcoin
      file: bitcoin.png
    - id: door
      file: door.gif
    - id: doorbell
      file: doorbell.gif
    - id: garage
      file: garage.gif
    - id: girl
      file: girl.gif
      
# Connection status and buttons exposed to HA
binary_sensor:
  - platform: status
    name: "Clock Connection Status"
    id: clock_connection_status
  - platform: gpio
    name: "$devicename left button"
    icon: "mdi:menu-left-outline"
    pin:
      number: $left_button_pin
      inverted: true
    on_press:
      then:
        - number.decrement: screen_brightness
  - platform: gpio
    name: "$devicename middle button"
    icon: "mdi:circle-outline"
    pin: 
      number: $mid_button_pin
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: displaycontrol
  - platform: gpio
    name: "$devicename right button"
    icon: "mdi:menu-right-outline"
    pin: 
      number: $right_button_pin
      inverted: true
    on_press:
      then:
        - number.increment: screen_brightness

logger:
  level: INFO

# Services exposed to HA
api:
  services:
    - service: alarm
      variables:
        icon_name: string
        text: string
      then:
        lambda: |-
          id(rgb8x32)->add_screen(icon_name,text,7,20,true);
          id(rgb8x32)->force_screen(icon_name);
    - service: brightness
      variables:
        brightness: int
      then:
        lambda: |-
          id(rgb8x32)->set_brightness(brightness);
    - service: icons
      then:
        lambda: |-
          id(rgb8x32)->show_all_icons();
    - service: skip_screen
      then:
        lambda: |-
          id(rgb8x32)->skip_screen();
    - service: tuneplay
      variables:
        tune: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return tune;'

# Gets/sets the brightness (0-255)
number:
  - platform: template
    name: "$devicename brightness"
    id: screen_brightness
    icon: "mdi:brightness-5"
    min_value: 0
    max_value: 255
    update_interval: 1s
    step: 25
    lambda: |-
      return id(rgb8x32)->get_brightness();
    set_action:
      lambda: |-
        id(rgb8x32)->set_brightness(x);

# Turns display on/off
switch:
  - platform: template
    name: "$devicename Display"
    id: displaycontrol
    icon: "mdi:power"
    restore_mode: ALWAYS_ON
    lambda: |-
      return id(rgb8x32)->show_display;
    turn_on_action:
      lambda: |-
        id(rgb8x32)->set_display_on();
    turn_off_action:
      lambda: |-
        id(rgb8x32)->set_display_off();

sensor:
  - platform: adc
    id: light_sensor
    name: "$devicename Illuminance"
    pin: $ldr_pin
    update_interval: 10s
    attenuation: auto
    device_class: illuminance
    unit_of_measurement: lx
    accuracy_decimals: 0
    filters:
      - lambda: |-
          return (x / 10000.0) * 2000000.0 - 15 ;

# Speaker: works together with rtttl
output:
  - platform: ledc
    pin: $buzzer_pin
    id: rtttl_out

# Tone generator: works together with output
rtttl:
  output: rtttl_out

# I2C Comms: works together with time
i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

# Clock sync: works together with i2c
time:
  - platform: homeassistant
    on_time_sync:
      then:
        ds1307.write_time:
  - platform: ds1307
    update_interval: never
    id: ehmtx_time

# Light Comms: works together with display
light:
  - platform: neopixelbus
    id: ehmtx_light
    internal: true #Hide from HA
    type: GRB
    variant: WS2812
    pin: $matrix_pin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

# Display Matrix: works together with light
display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();
```

nukefrenzy, 2023-04-21:
For now I'm getting around this by calling the following actions after I call "add_screen".  It waits for the screen I just added to be displayed and immediately deletes it.  It times out and aborts if it doesn't see the screen after 1 minute.

```
wait_for_trigger:
  - platform: event
    event_data:
      iconname: doorbell
    event_type: esphome.next_screen
timeout:
  hours: 0
  minutes: 1
  seconds: 0
  milliseconds: 0
continue_on_timeout: false
service: esphome.ulanzi_del_screen
data:
  icon_name: doorbell
```

lubeda, 2023-04-22:
_This is not what's happening. Instead, it just keeps showing the doorbell screen over and over until it expires._

This is right, thats by design!  In the upcoming version [EspHoMaTriXv2](https://github.com/lubeda/EspHoMaTriXv2) you will have more control, so you can add blank screens in the loop. The old version will stay as it is now, perhaps some bugfixes. The new version is not stable enough for "production" yet.

nukefrenzy, 2023-04-22:
The graphic on the readme is confusing then.  It shows each screen being displayed one time and then going to the clock/date before showing the added screens again.  So it seems like a full rotation includes the clock every time.

lubeda, 2023-05-14:
Hi, the [new](https://github.com/lubeda/EsphoMaTrixv2/) version gives away to better control the clock/date display. I will not fix this in this version!

closed 2023-05-14 by lubeda