lubeda/EsphoMaTrix #100: Display problem since update

bobdivx, 2023-04-22:
Hello, a few months ago I installed your project on my esp 8266, today I only have two lines left to display.
My code worked perfectly, today impossible to update.

here is my code:
`substitutions:
  devicename: esphomatrix
  mypin: GPIO16
  board: mhetesp32minikit

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EsphoMaTrix

esphome:
  name: $devicename
  on_boot:
    priority: -100
    then: 
      - ehmtx.text.color:
          id: rgb8x32
          red: !lambda return 200;
          blue: !lambda return 170;
      - ehmtx.today.color:
          id: rgb8x32
          red: !lambda return 10;
          green: !lambda return 250;
      - ehmtx.clock.color:
          id: rgb8x32
          red: !lambda return 50;
          green: !lambda return 150;
          blue: !lambda return 230;
      - ehmtx.weekday.color:
          id: rgb8x32
          red: !lambda return 250;
          green: !lambda return 50;
          blue: !lambda return 30;

esp32:
  board: $board
  

binary_sensor:
  - platform: status
    name: "$devicename Status"

font:
  - file: fonts/monobit.ttf
    id: ehmtx_font
    size: 16
    glyphs:  |
      !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@<>/
logger:

# Enable Home Assistant API
api:
  services:
    - service: alarm
      variables:
        icon_name: string
        text: string
      then:
        lambda: |-
          id(rgb8x32)->add_screen(icon_name, text, 7, true);
          id(rgb8x32)->force_screen(icon_name);
    - service: screen
      variables:
        icon_name: string
        text: string
      then:
[... 284 lines cut ...]
    - url: https://developer.lametric.com/content/apps/icon_thumbs/16935
      id: mario
      duration: 250
    - url: https://developer.lametric.com/content/apps/icon_thumbs/16587
      id: wash
      duration: 250
    - url: https://developer.lametric.com/content/apps/icon_thumbs/24120
      id: tank100
      duration: 250
    - url: https://developer.lametric.com/content/apps/icon_thumbs/2361
      id: play
      duration: 250
    - url: https://developer.lametric.com/content/apps/icon_thumbs/39039
      id: temp_outside
      duration: 250
    - url: https://developer.lametric.com/content/apps/icon_thumbs/20944
      id: paused
select.ehmtx: [source /config/esphome/esphomatrix.yaml:262]
  
  Platform not found: 'select.ehmtx'.
  platform: ehmtx
  id: esphomatrix
  name: esphomatrix
required key not provided
`

lubeda, 2023-04-22:
Did you read  https://github.com/lubeda/EsphoMaTrix/#how-to-update ?

bobdivx, 2023-04-22:
yes I saw that, but I confess that I am a little lost, attached my file to modify but I think I missed something
[matrix.txt](https://github.com/lubeda/EsphoMaTrix/files/11301769/matrix.txt)

lubeda, 2023-04-22:
[bobdivx.txt](https://github.com/lubeda/EsphoMaTrix/files/11301783/bobdivx.txt)

closed 2023-04-22 by lubeda

bobdivx, 2023-04-22:
thank you very much for your time, when I want to compile I have this error

`/config/esphome/esphomatrix.yaml: In lambda function:
/config/esphome/esphomatrix.yaml:60:51: error: no matching function for call to 'esphome::EHMTX::add_screen(std::__cxx11::string&, std::__cxx11::string&, int, bool)'
           id(rgb8x32)->add_screen(icon_name, text, 7, true);
                                                   ^
In file included from src/esphome.h:21,
                 from src/main.cpp:3:
src/esphome/components/ehmtx/EHMTX.h:95:10: note: candidate: 'void esphome::EHMTX::add_screen(std::__cxx11::string, std::__cxx11::string, int, int, bool)'
     void add_screen(std::string icon, std::string text, int duration, int showt_time, bool alarm);
          ^~~~~~~~~~
src/esphome/components/ehmtx/EHMTX.h:95:10: note:   candidate expects 5 arguments, 4 provided
/config/esphome/esphomatrix.yaml: In lambda function:
/config/esphome/esphomatrix.yaml:68:52: error: no matching function for call to 'esphome::EHMTX::add_screen(std::__cxx11::string&, std::__cxx11::string&, int, bool)'
           id(rgb8x32)->add_screen(icon_name, text, 1, false);
                                                    ^
In file included from src/esphome.h:21,
                 from src/main.cpp:3:
src/esphome/components/ehmtx/EHMTX.h:95:10: note: candidate: 'void esphome::EHMTX::add_screen(std::__cxx11::string, std::__cxx11::string, int, int, bool)'
     void add_screen(std::string icon, std::string text, int duration, int showt_time, bool alarm);
          ^~~~~~~~~~
src/esphome/components/ehmtx/EHMTX.h:95:10: note:   candidate expects 5 arguments, 4 provided
/config/esphome/esphomatrix.yaml: In lambda function:
/config/esphome/esphomatrix.yaml:92:16: error: 'class esphome::EHMTX' has no member named 'set_indicator_color'; did you mean 'indicator_color'?
           id(rgb8x32)->set_indicator_color(r,g,b);
                ^~~~~~~~~~~~~~~~~~~
                indicator_color
/config/esphome/esphomatrix.yaml:93:33: error: no matching function for call to 'esphome::EHMTX::set_indicator_on()'
           id(rgb8x32)->set_indicator_on();
                                 ^
In file included from src/esphome.h:21,
                 from src/main.cpp:3:
src/esphome/components/ehmtx/EHMTX.h:106:10: note: candidate: 'void esphome::EHMTX::set_indicator_on(int, int, int)'
     void set_indicator_on(int r, int g, int b);
          ^~~~~~~~~~~~~~~~
src/esphome/components/ehmtx/EHMTX.h:106:10: note:   candidate expects 3 arguments, 0 provided
/config/esphome/esphomatrix.yaml: In lambda function:
/config/esphome/esphomatrix.yaml:117:27: error: 'icon_name' was not declared in this scope
             id(rgb8x32)->add_screen(icon_name,text,7,30,true);
                           ^~~~~~~~~
/config/esphome/esphomatrix.yaml:117:37: error: 'text' was not declared in this scope
             id(rgb8x32)->add_screen(icon_name,text,7,30,true);
                                     ^~~~
/config/esphome/esphomatrix.yaml:117:37: note: suggested alternative: 'atexit'
             id(rgb8x32)->add_screen(icon_name,text,7,30,true);
                                     ^~~~
                                     atexit
Compiling /data/esphomatrix/.pioenvs/esphomatrix/lib64d/WiFi/WiFiUdp.cpp.o
Compiling /data/esphomatrix/.pioenvs/esphomatrix/lib915/ESPmDNS/ESPmDNS.cpp.o
Compiling /data/esphomatrix/.pioenvs/esphomatrix/libbc6/Update/HttpsOTAUpdate.cpp.o
Compiling /data/esphomatrix/.pioenvs/esphomatrix/libbc6/Update/Updater.cpp.o
*** [/data/esphomatrix/.pioenvs/esphomatrix/src/main.cpp.o] Error 1
========================= [FAILED] Took 12.73 seconds =========================`

bobdivx, 2023-04-22:
I have modified the lines as in your cedpanat documentation, I only have one error left:
esphome: $devicename

[esphomatrix.yaml.txt](https://github.com/lubeda/EsphoMaTrix/files/11302072/esphomatrix.yaml.txt)