lubeda/EsphoMaTrix #106: force_screen when time/date screen is active

hugospacep, 2023-06-03:
if force_screen is sent while time/date screen is displaying, 'time/date screen' seems to be restarted. Then instead of forcing the  'forced screen' as needed, it is added to the end of the queue of other screens.

ULANZI TC001

lubeda, 2023-06-04:
Can you provide a debug log by setting:

```
logger:
  level: DEBUG
```

in your yaml file?

hugospacep, 2023-06-04:
Of course, here is the log:

> [14:56:28][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:56:28][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:56:28][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:29][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:30][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:30][D][sensor:094]: 'ulanzi Battery': Sending state 106.86387 % with 0 decimals of accuracy
[14:56:31][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:32][I][ota:113]: Boot seems successful, resetting boot loop counter.
[14:56:32][D][esp32.preferences:114]: Saving 1 preferences to flash...
[14:56:32][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[14:56:32][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:33][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:34][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:35][D][EHMTX:293]: next action: show clock/date for 6/2 sec
[14:56:35][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:36][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:36][D][EHMTX:094]: display on
**[14:56:36][D][EHMTX:022]: delete screen icon: 9**
**[14:56:36][D][EHMTX:151]: icon: l_alarme_presence found id: 11
[14:56:36][D][EHMTX:016]: findfreeslot for icon: 11
[14:56:36][D][EHMTX:136]: display length text: ... pixels 18 calculated: 30 show_time: 30 default: 8
[14:56:36][D][EHMTX:419]: add_screen icon: 11 iconname: l_alarme_presence text: ... lifetime: 1 screen_time: 30 alarm: 0
[14:56:36][D][EHMTX:151]: icon: l_alarme_presence found id: 11
[14:56:36][D][EHMTX:032]: force next screen: l_alarme_presence for 8 sec
[14:56:37][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:38][D][sensor:094]: 'ulanzi Illuminance': Sending state 512.40460 lx with 0 decimals of accuracy
[14:56:38][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:39][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:40][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:41][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:42][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:43][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:44][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:45][D][EHMTX:305]: next action: show screen "l_radio_2" for 15 sec
[14:56:45][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:46][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:47][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:48][D][sensor:094]: 'ulanzi Illuminance': Sending state 512.20477 lx with 0 decimals of accuracy
[14:56:48][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:49][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:50][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:51][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:52][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:53][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:54][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:55][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:56][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:57][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:58][D][sensor:094]: 'ulanzi Illuminance': Sending state 512.80420 lx with 0 decimals of accuracy
[14:56:58][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:56:58][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:56:58][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:56:59][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:00][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:01][D][EHMTX:305]: next action: show screen "l_alarme_presence" for 30 sec
[14:57:01][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:02][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:03][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:04][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:05][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:06][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:07][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:08][D][sensor:094]: 'ulanzi Illuminance': Sending state 470.04596 lx with 0 decimals of accuracy
[14:57:08][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:57:08][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:57:08][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:09][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:10][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[... 15 lines cut ...]
[14:57:23][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:24][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:25][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:26][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:27][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[... cut ...]
[14:57:35][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:36][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:37][D][number:012]: 'ulanzi brightness': Sending state 150.000000
[14:57:38][D][sensor:094]: 'ulanzi Illuminance': Sending state 472.64343 lx with 0 decimals of accuracy
[14:57:38][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:57:38][I][EHMTX:496]: set_brightness 150 => 58.82 %
[14:57:38][D][number:012]: 'ulanzi brightness': Sending state 150.000000
>
Thanks

lubeda, 2023-06-04:
OK, I see the problem. 

Can you provide another log with a call of get_status immediately before and after the force_screen service call?

hugospacep, 2023-06-04:
Ok, here is the log with get status:

> [19:16:25][D][EHMTX:305]: next action: show screen "l_radio_2" for 15 sec
[19:16:27][D][sensor:094]: 'ulanzi Illuminance': Sending state 389.92419 lx with 0 decimals of accuracy
[19:16:29][D][number:012]: 'ulanzi brightness': Sending state 100.000000
[19:16:37][D][sensor:094]: 'ulanzi Illuminance': Sending state 388.72537 lx with 0 decimals of accuracy
[19:16:37][D][number:054]: 'ulanzi brightness' - Setting number value
[19:16:37][D][number:113]:   New number value: 100.000000
[19:16:37][I][EHMTX:496]: set_brightness 100 => 39.22 %
[19:16:39][D][number:012]: 'ulanzi brightness': Sending state 100.000000
[19:16:41][D][EHMTX:293]: next action: show clock/date for 6/2 sec
[19:16:42][D][EHMTX:094]: display on
[19:16:42][D][EHMTX:022]: delete screen icon: 11
[19:16:43][D][EHMTX:151]: icon: l_alarme_presence found id: 11
[19:16:43][D][EHMTX:016]: findfreeslot for icon: 11
[19:16:43][D][EHMTX:136]: display length text: ... pixels 18 calculated: 30 show_time: 30 default: 8
[19:16:43][D][EHMTX:419]: add_screen icon: 11 iconname: l_alarme_presence text: ... lifetime: 1 screen_time: 30 alarm: 0
[19:16:43][I][EHMTX:342]: status time: 4.6.2023 19:16
[19:16:43][I][EHMTX:343]: status brightness: 100 (0..255)
[19:16:43][I][EHMTX:344]: status date format: %d-%m
[19:16:43][I][EHMTX:345]: status time format: %H:%M
[19:16:43][I][EHMTX:346]: status text_color: RGB(240,240,240)
[19:16:43][I][EHMTX:347]: status alarm_color: RGB(200,50,50)
[19:16:43][I][EHMTX:354]: status indicator off
[19:16:43][I][EHMTX:358]: status display on
[19:16:43][I][EHMTX:150]: status active slot: 0
[19:16:43][I][EHMTX:151]: status screen count: 2 of 24
[19:16:43][I][EHMTX:158]: status slot 0 icon 8 text: Jazz alarm: 0 dd: 15 sec end: 3473 sec
[19:16:43][I][EHMTX:158]: status slot 1 icon 11 text: ... alarm: 0 dd: 30 sec end: 60 sec
[19:16:43][I][EHMTX:369]: status icon: 0 name: ha
[19:16:43][I][EHMTX:369]: status icon: 1 name: tempc
[19:16:43][I][EHMTX:369]: status icon: 2 name: plug
[19:16:43][I][EHMTX:369]: status icon: 3 name: humidity
[19:16:43][I][EHMTX:369]: status icon: 4 name: co2
[19:16:43][I][EHMTX:369]: status icon: 5 name: alarme
[19:16:43][I][EHMTX:369]: status icon: 6 name: radio
[19:16:43][I][EHMTX:369]: status icon: 7 name: l_radio
[19:16:43][I][EHMTX:369]: status icon: 8 name: l_radio_2
[19:16:43][I][EHMTX:369]: status icon: 9 name: l_alarme_armee
[19:16:43][I][EHMTX:369]: status icon: 10 name: l_alarme_armement
[19:16:43][I][EHMTX:369]: status icon: 11 name: l_alarme_presence
[19:16:43][I][EHMTX:369]: status icon: 12 name: l_k2000
[19:16:43][I][EHMTX:369]: status icon: 13 name: l_code_rouge
[19:16:43][I][EHMTX:369]: status icon: 14 name: l_alerte_rouge
[19:16:43][I][EHMTX:369]: status icon: 15 name: l_alerte_orange
[19:16:43][I][EHMTX:369]: status icon: 16 name: l_alerte_jaune
**[19:16:43][D][EHMTX:151]: icon: l_alarme_presence found id: 11
[19:16:43][D][EHMTX:032]: force next screen: l_alarme_presence for 8 sec**
[19:16:43][I][EHMTX:342]: status time: 4.6.2023 19:16
[19:16:43][I][EHMTX:343]: status brightness: 100 (0..255)
[19:16:43][I][EHMTX:344]: status date format: %d-%m
[19:16:43][I][EHMTX:345]: status time format: %H:%M
[19:16:43][I][EHMTX:346]: status text_color: RGB(240,240,240)
[19:16:43][I][EHMTX:347]: status alarm_color: RGB(200,50,50)
[19:16:43][I][EHMTX:354]: status indicator off
[19:16:43][I][EHMTX:358]: status display on
[19:16:43][I][EHMTX:150]: status active slot: 1
[19:16:43][I][EHMTX:151]: status screen count: 2 of 24
[19:16:43][I][EHMTX:158]: status slot 0 icon 8 text: Jazz alarm: 0 dd: 15 sec end: 3473 sec
[19:16:43][I][EHMTX:158]: status slot 1 icon 11 text: ... alarm: 0 dd: 30 sec end: 60 sec
[19:16:43][I][EHMTX:369]: status icon: 0 name: ha
[19:16:43][I][EHMTX:369]: status icon: 1 name: tempc
[19:16:43][I][EHMTX:369]: status icon: 2 name: plug
[19:16:43][I][EHMTX:369]: status icon: 3 name: humidity
[19:16:43][I][EHMTX:369]: status icon: 4 name: co2
[19:16:43][I][EHMTX:369]: status icon: 5 name: alarme
[19:16:43][I][EHMTX:369]: status icon: 6 name: radio
[19:16:43][I][EHMTX:369]: status icon: 7 name: l_radio
[19:16:43][I][EHMTX:369]: status icon: 8 name: l_radio_2
[19:16:43][I][EHMTX:369]: status icon: 9 name: l_alarme_armee
[... 13 lines cut ...]
[19:16:57][D][sensor:094]: 'ulanzi Illuminance': Sending state 390.12399 lx with 0 decimals of accuracy
[19:16:59][D][number:012]: 'ulanzi brightness': Sending state 100.000000
[19:17:07][D][sensor:094]: 'ulanzi Illuminance': Sending state 391.12302 lx with 0 decimals of accuracy
[19:17:07][D][number:054]: 'ulanzi brightness' - Setting number value
[19:17:07][D][number:113]:   New number value: 100.000000
[19:17:07][I][EHMTX:496]: set_brightness 100 => 39.22 %
[19:17:07][D][EHMTX:305]: next action: show screen "l_alarme_presence" for 30 sec
[19:17:09][D][number:012]: 'ulanzi brightness': Sending state 100.000000
[19:17:17][D][sensor:094]: 'ulanzi Illuminance': Sending state 391.72244 lx with 0 decimals of accuracy
[19:17:17][D][number:054]: 'ulanzi brightness' - Setting number value
[19:17:17][D][number:113]:   New number value: 100.000000
[19:17:17][I][EHMTX:496]: set_brightness 100 => 39.22 %
[19:17:19][D][number:012]: 'ulanzi brightness': Sending state 100.000000
[19:17:27][D][sensor:094]: 'ulanzi Illuminance': Sending state 391.12302 lx with 0 decimals of accuracy
[19:17:27][D][number:054]: 'ulanzi brightness' - Setting number value
[19:17:27][D][number:113]:   New number value: 100.000000
[... cut ...]
[19:17:29][D][number:012]: 'ulanzi brightness': Sending state 100.000000
[19:17:37][D][sensor:094]: 'ulanzi Illuminance': Sending state 391.92224 lx with 0 decimals of accuracy
[19:17:37][D][number:054]: 'ulanzi brightness' - Setting number value
[19:17:37][D][number:113]:   New number value: 100.000000
[19:17:37][I][EHMTX:496]: set_brightness 100 => 39.22 %
[19:17:38][D][EHMTX:305]: next action: show screen "l_radio_2" for 15 sec
[19:17:39][D][number:012]: 'ulanzi brightness': Sending state 100.000000
>

lubeda, 2023-10-31:
This repo will be archived, see https://github.com/lubeda/EsphoMaTrixv2

closed 2023-10-31 by lubeda