lubeda/EsphoMaTrix #113: Change readme to include Ulanzi TC001 mods

popy2k14, 2023-06-24:
I have made hardwarte mods to the TC001:

- remove battery
- always on mode

See here: https://www.reddit.com/r/homeassistant/comments/131okf7/comment/jp0e1tw/?utm_source=share&utm_medium=web2x&context=3

Is it possible to add this to the readme so other can also benefit from this?

larsxschneider, 2023-06-28:
@popy2k14 I am curious, why did you remove the battery? I like my TC001 to be always on too and I don't like to have yet another battery (🔥 !) in my home but I am also frightened that I would destroy the TC001 in the process 😄

popy2k14, 2023-06-28:
@larsxschneider That's exactly the reason.
The clock is always on , and i don't want to have another battery in my home (🔥!).
With the HW mods from the link above, it's working just with the USB power supply.
The process is'nt challenging, just use an opener tool (plastic) to open gently pull the cover and follow my guide step by step.

PS.: i am switching the display on/off when someone is in the room or leaves it). WIth this i am saving a ton of power.

larsxschneider, 2023-06-28:
Thank you! How much power does the device draw when it is on? How did you figure out _where_ to solder the wire?  (I have no clue about hardware 😄)

lubeda, 2023-10-31:
This repo will be archived, see https://github.com/lubeda/EsphoMaTrixv2

popy2k14, 2023-10-31:
@lubeda can you please add the info to the V2 repo? Thx

lubeda, 2023-11-01:
its in the wiki now