lubeda/EsphoMaTrix #115: ESPHome 2023.7.0 font breaking change

navet, 2023-07-19:
The latest version of ESPHome 2023.7.0 has some [breaking changes](https://github.com/esphome/esphome/pull/4963) with font 

Trying to update to latest version causes the following error.

`INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/esphome-web-7ca7a8.yaml...
INFO Updating https://github.com/lubeda/EsphoMaTrix@None
WARNING 
WARNING If you are upgrading EsphoMaTrix from a version before 2023.4.0,
WARNING you should read the section https://github.com/lubeda/EsphoMaTrix/#how-to-update for tipps.
WARNING 
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/London'
INFO Detected timezone 'Europe/London'
INFO Generating C++ source...
INFO Compiling app...
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.4
|-- NeoPixelBus @ 2.7.3
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX.cpp.o
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_icons.cpp.o
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtx/EHMTX.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:59:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *font;
              ^~~~
src/esphome/components/ehmtx/EHMTX.h:99:28: error: 'esphome::display::Font' has not been declared
     void set_font(display::Font *font);
                            ^~~~
src/esphome/components/ehmtx/EHMTX.h:479:3: error: expected class-name before '{' token
   {
   ^
src/esphome/components/ehmtx/EHMTX.h:484:107: error: 'esphome::display::ImageType' has not been declared
     EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration);
                                                                                                           ^~~~~~~~~
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtx/EHMTX_icons.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:59:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *font;
              ^~~~
src/esphome/components/ehmtx/EHMTX.h:99:28: error: 'esphome::display::Font' has not been declared
     void set_font(display::Font *font);
                            ^~~~
src/esphome/components/ehmtx/EHMTX.h:479:3: error: expected class-name before '{' token
   {
   ^
src/esphome/components/ehmtx/EHMTX.h:484:107: error: 'esphome::display::ImageType' has not been declared
     EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration);
                                                                                                           ^~~~~~~~~
src/esphome/components/ehmtx/EHMTX.cpp: In member function 'void esphome::EHMTX::draw_clock()':
src/esphome/components/ehmtx/EHMTX.cpp:183:74: error: 'class esphome::EHMTX' has no member named 'font'
         this->display->strftime(this->xoffset + 15, this->yoffset, this->font, this->clock_color, display::TextAlign::BASELINE_CENTER, this->time_fmt.c_str(),
                                                                          ^~~~
src/esphome/components/ehmtx/EHMTX.cpp:192:74: error: 'class esphome::EHMTX' has no member named 'font'
         this->display->strftime(this->xoffset + 15, this->yoffset, this->font, this->clock_color, display::TextAlign::BASELINE_CENTER, this->date_fmt.c_str(),
                                                                          ^~~~
src/esphome/components/ehmtx/EHMTX.cpp:199:69: error: 'class esphome::EHMTX' has no member named 'font'
[... 33 lines cut ...]
                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtx/EHMTX_icons.cpp:26:9: error: 'Animation' has not been declared
         Animation::next_frame();
         ^~~~~~~~~
src/esphome/components/ehmtx/EHMTX_icons.cpp:30:19: error: 'class esphome::EHMTX_Icon' has no member named 'get_current_frame'
         if (this->get_current_frame() == 1) // this->get_animation_frame_count())
                   ^~~~~~~~~~~~~~~~~
src/esphome/components/ehmtx/EHMTX_icons.cpp:34:9: error: 'Animation' has not been declared
         Animation::prev_frame();
         ^~~~~~~~~
src/esphome/components/ehmtx/EHMTX.cpp:373:43: error: expected primary-expression before ')' token
   void EHMTX::set_font(display::Font *font)
                                           ^
src/esphome/components/ehmtx/EHMTX.cpp: In member function 'void esphome::EHMTX::internal_add_screen(uint8_t, std::__cxx11::string, uint16_t, uint16_t, bool)':
src/esphome/components/ehmtx/EHMTX.cpp:432:62: error: 'class esphome::EHMTX' has no member named 'font'
     this->display->get_text_bounds(0, 0, text.c_str(), this->font, display::TextAlign::LEFT, &x, &y, &w, &h);
                                                              ^~~~
src/esphome/components/ehmtx/EHMTX.cpp: In member function 'void esphome::EHMTX::show_all_icons()':
src/esphome/components/ehmtx/EHMTX.cpp:607:78: error: 'class esphome::EHMTX' has no member named 'font'
     this->display->get_text_bounds(0, 0, this->icons[0]->name.c_str(), this->font, display::TextAlign::LEFT, &x, &y, &w, &h);
                                                                              ^~~~
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_icons.cpp.o] Error 1
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX.cpp.o] Error 1
========================== [FAILED] Took 4.17 seconds ==========================
`

scarolan, 2023-07-21:
similar errors here:

```
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.4
|-- NeoPixelBus @ 2.7.3
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX.cpp.o
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_icons.cpp.o
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtx/EHMTX.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:6:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX.cpp.o] Error 1
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtx/EHMTX_icons.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:6:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_icons.cpp.o] Error 1
========================== [FAILED] Took 2.66 seconds ==========================
```

pepe59, 2023-07-21:
It is the same here with version ESPHome 2023.7

navet, 2023-07-22:
Fixed it by adding to the yaml per https://github.com/lubeda/EspHoMaTriXv2/issues/62

Also didnt realize there was a v2, so i ported over to that as well. 
```
image:
   - file: 1pixel.gif
     id: breaking20237

animation:
  - file: 1pixel.gif
    id: breaking202371
```

closed 2023-07-22 by navet