lubeda/EsphoMaTrix #116: animation.h is missing

UlrichThiess, 2023-07-25:
```INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/ulanzi-a.yaml...
WARNING 
WARNING If you are upgrading EsphoMaTrix from a version before 2023.4.0,
WARNING you should read the section https://github.com/lubeda/EsphoMaTrix/#how-to-update for tipps.
WARNING This version of EsphoMaTrix works only on esphome > 2023.7.0
WARNING 
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing ulanzi-a (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/adc/adc_sensor.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/addressable_light/addressable_light_display.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/api_connection.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/api_server.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/list_entities.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/proto.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/api/user_services.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/display/display.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/display/display_buffer.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/display/rect.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/ds1307/ds1307.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/ehmtx/EHMTX.cpp.o
Compiling /data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/ehmtx/EHMTX_icons.cpp.o
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtx/EHMTX_icons.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:6:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtx/EHMTX.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:6:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/ehmtx/EHMTX_icons.cpp.o] Error 1
*** [/data/ulanzi-a/.pioenvs/ulanzi-a/src/esphome/components/ehmtx/EHMTX.cpp.o] Error 1
========================= [FAILED] Took 12.29 seconds =========================```

lubeda, 2023-07-26:
Take a look at the README.md, there is a workaround

closed 2023-07-26 by lubeda

Fiiti, 2023-08-25:
Same to me.

Please excuse the silly question, but it is not clear to me what exactly animation.h does. 
Thus, unfortunately, it is also not clear to me what exactly the workaround should be.

Please be so kind and tell me under which point in the long read.me the solution/workaround is.

Regards.

lubeda, 2023-08-25:
Here: 
https://github.com/lubeda/EspHoMaTriXv2/issues/62#issuecomment-1643052894