lubeda/EsphoMaTrix #121: Screen with Color

michisbarq, 2023-10-24:
Hi,
I add a Screen in HA and I want this Screen in a specific Text Color.

Actual I call to Services. Add_Screen and Text_Color.

That doesn't work very fine. Sometimes it work, sometimes not!

Is it possible to Add_Screen with a specific COLOR in RGB?

lubeda, 2023-10-25:
Hi, take a look at https://github.com/lubeda/EspHoMaTriXv2 in the 2023.9.1 branch. This is the evolution of your version, so you can expect more functionality but also the need of changing your service calls. e.g.

here you can set the color per icon: 
icon_screen => {"icon_name", "text", "lifetime", "screen_time", "default_font", "r", "g", "b"}

michisbarq, 2023-10-25:
Thank U. I try it with my new Ulanzi Clock.

closed 2023-10-25 by michisbarq